RobTillaart/HX711 #1: Persistent calibration?

adamcoulombe, 2020-09-20:
Is there any way to store calibration settings so that you dont have to reset the zero point when the controller is powered on each time?

adamcoulombe, 2020-09-20:
Nevermind! I read thru some of the issues in the bogde version of this lib and found my answer.

Basically I what i was looking for was the offset (basically what the `tare` function sets).

You can read the offset using `Serial.println(scale.get_offset())` and use this value rather than using tare every time. You can just use `scale.get_offset(8266896)` (or whatever value you need to use)

RobTillaart, 2020-09-21:
The underlying formula is  weight  =  (raw read - offset) / scale

To store the calibration data you need to call both ```float get_scale()``` and  ```long get_offset()``` and store both of them.
They need 4 bytes of storage each normally you need 8 bytes per loadcell.

closed 2020-09-21 by RobTillaart

ClemensGruber, 2021-05-20:
@adamcoulombe you can use e.g. this sketch to get the values and the code you have to use in your "persistant" sketch. Finally you have to set 

```
scale.set_offset(xx);
scale.set_scale(yy.zz);

```
with your values to get a persistant output based on a one time calibration that survives a reset also.  

```
#include "HX711.h"

HX711 scale;
uint8_t dataPin = 2;
uint8_t clockPin = 13;

void setup()
{
  Serial.begin(115200);
  scale.begin(dataPin, clockPin);

  Serial.print("UNITS: ");
  Serial.println(scale.get_units(10));
  
  Serial.println("\nEmpty the scale, press a key to continue");
  while(!Serial.available());
  while(Serial.available()) Serial.read();
  
  scale.tare();
  Serial.print("UNITS: ");
  Serial.println(scale.get_units(10));

  Serial.println("\nPut 1000 gram in the scale, press a key to continue");
  while(!Serial.available());
  while(Serial.available()) Serial.read();

  scale.calibrate_scale(1000, 5);
  Serial.print("UNITS: ");
  Serial.println(scale.get_units(10));

  Serial.println("\nScale is calibrated, your calibration values:");

  long scaleOffset = scale.get_offset();
  Serial.print("\nOffset \t");
  Serial.println(scaleOffset);
  
  float scaleFactor = scale.get_scale();
  Serial.print("Scale \t");
  Serial.println(scaleFactor);

  Serial.println("\nUse this code for setting zero and calibration factor permanently:");

  Serial.print("\nscale.set_offset(");
  Serial.print(scaleOffset);
  Serial.println(");"); 
  Serial.print("scale.set_scale(");
  Serial.print(scaleFactor);
  Serial.println(");"); 

  Serial.println("\nPress a key to continue");
  while(!Serial.available());
  while(Serial.available()) Serial.read();
}

void loop()
{
  Serial.print("UNITS: ");
  Serial.println(scale.get_units(10));
  delay(250);
}

```