RobTillaart/HX711 #6: Typo in function name cal_l_ibrate_scale(uint16_t

ClemensGruber, 2021-05-11:
Andreas / @amotl pointed me via https://community.hiveeyes.org/t/neue-hx711-lib-von-rob-tillaart/3994/2 to a typo in the function name `callibrate_scale()`, should be one "l" instead off "ll", e.g.

https://github.com/RobTillaart/HX711/blob/51da457921774fc9cf37d2fd9a03d4910b592b96/HX711.cpp#L89

https://github.com/RobTillaart/HX711/blob/51da457921774fc9cf37d2fd9a03d4910b592b96/HX711.h#L74
https://github.com/RobTillaart/HX711/blob/51da457921774fc9cf37d2fd9a03d4910b592b96/HX711.h#L76

Maybe the other remark about `shiftInSlow() `is also interesting. I'm not so deep involved in the HX711 mechanism so just a hint to the discussion from my side without further technical substance. ;-)

RobTillaart, 2021-05-11:
calibrate is indeed with one L 
A fix will be made later today.

RobTillaart, 2021-05-11:
typo will be fixed in the read_median branch.

RobTillaart, 2021-05-16:
solved in 0.2.2

closed 2021-05-16 by RobTillaart