RobTillaart/HX711 #13: Load cell liquid filling

Oceanlig, 2021-12-13:


RobTillaart, 2021-12-13:
@Oceanlig   issue is moved to the HX711 library's own repo.

RobTillaart, 2021-12-13:
>  Like if I need to fill 10kg of liquid, ...

Than you need loadcells that add up to 15-20kg. Given that max precision is 0.1~1% would mean  10~100 grams or so.

That said it might be possible to use only one load cell and use pulley's, hinges and gears to scale down the forces of weight.

There are other ways too, filling a bottle can be monitored by 
- a distance sensor that detects when a contra-weight is in balance or starts to move. 
- a light beam that 'breaks' when bottle is filled. (light beam goes only through glass or through glass and liquid)
- time, if your flow is constant this can work pretty well. Should definitely be used as "safety net" for when other method fails.
  e.g. the filling takes normally 58-62 seconds, then a timer should stop the thing after 63 seconds. 
-

RobTillaart, 2021-12-19:
As there are no further questions I close this issue. 
Feel free to reopen it if you have more questions in the same context.

closed 2021-12-19 by RobTillaart