RobTillaart/HX711 #16: Error when using with https://create.arduino.cc/

dark7solutions, 2022-03-16:
When using the library with Arduino cloud web editor I get the following error:

```
fatal error: HX711.h : No such file or directory

#include <HX711.h >
```

Removing the space after the .h seems to fix the problem. I think the problem lies in the library.properties where there is an extra space follwing the includes=HX711.h on line 10 but I'm still learning how these files function in the online editor

RobTillaart, 2022-03-16:
Thanks for this issue,
Never tested in that environment, will investigate later this week.

RobTillaart, 2022-03-16:
There will be a minor update (0.3.2)  today and I will remove the trailing space too. 
If the problem still exists when using 0.3.2 more investigation is needed.

dark7solutions, 2022-03-16:
Just tested it on the web editor and it seems like it works! Thanks so much.

RobTillaart, 2022-03-16:
You're welcome