RobTillaart/HX711 #20: Possible to make this library 'async'?

TWCCarlson, 2022-04-12:
I see that there is an `is_ready()` function in the .cpp. What I would like to do is request the HX711 to capture some data, handle acquisition from other sensors (using your ADS1115 library, actually) and then collect data from the HX711 at some point in the future when I am ready to package it all and send out over serial.

I didn't immediately see a function that does this, nor a call from the `read()` function to start a new conversion, though I suspect the loop decrementing `m` could be responsible.

I will dig into the HX711 datasheet more deeply and try to write this code myself unless I've simply missed it.

TWCCarlson, 2022-04-12:
Took a look at the timing diagram. It seems that once DOUT is driven high by the 25th pulse, only 0.1usec need to pass before PD_SCK can be driven high again for another conversion. I'm curious about the gain setting, you do it without a microsecond delay.

But I can see that I do not need to request a ready state to get a read.

closed 2022-04-12 by TWCCarlson

RobTillaart, 2022-04-12:
Think it is a good idea to make an example how to use the library in "async" way.
by checking ```is_ready();``` before calling ```read();``` the blocking part will be skipped.

Good question about "how a new read is triggered", I'll need to dive into the datasheet to have an answer.
It might be triggered by the internal ADC. If it is stable  the datapin goes LOW, so changing a weight /force will trigger conversion. Would make the most sense. Added a (low prio) todo for myself.