RobTillaart/HX711 #21: Creep and Temperature effect and removal. Load cell storage over long time

dev-techshlok, 2022-04-23:
Hello,
I am designing a load cell monitored storage. Where I can keep something for months to monitor the weight just like beehive monitor projects.
I am facing the creep effect after reading the common error document.
I have checked any wire r other issues everything is in place. Even I have used a 30Kg loadcell setup from a market product already. Just added HX711 with 4 wires.

My result is increasing over time by around 2 grams per 12-15 minutes.

Is there a way to compensate for this error in code?
Also is there any formula to compensate temperature?

RobTillaart, 2022-04-23:
Hi,
Compensation for temperature differs per load cell type. If you have a decent datasheet, these often give a graph for temperature error or a % per ° 
As the graph often is not lineair you might have a look at my multimap library/ function. 
A linear percentage correction is more straightforward to do in code.

The creep / drift is another type of problem. I know these occur especially if you use the load cell at the end of its working range. Think this can permanently damage the stiffness  of a device. So a rule of thumb is to use load cells which can handle at least 25% more than your max weight.

Does the creep stabilises e.g. after e.g 24 hr?

RobTillaart, 2022-04-23:
Some questions popped up.
Q: what is the weight you want to monitor?
Q: can the weight loss be caused (partlially) by evaporation or oxidation?

RobTillaart, 2022-05-08:
@dev-techshlok 
As no answers came upon my questions I assume the problem is solved so I close the issue.
Feel free to reopen if needed.

closed 2022-05-08 by RobTillaart