RobTillaart/HX711 #22: Problem with OLED 1.3"  SH1106

AlquiWKT, 2022-05-14:
Inglés

I have used a variation of an example from the library (HX_set_persistent). I have used an OLED SH1106 display. When I add it to the example the screen does not work. The screen works fine until I add the calibration block to it (although I don't enter the code)
[pruebaoled.zip](https://github.com/RobTillaart/HX711/files/8692788/pruebaoled.zip)
.

AlquiWKT, 2022-05-14:
I have tried another smaller OLED, an Adafruit_SSD1306 OLED, it works fine

RobTillaart, 2022-05-14:
Maybe you are running out of RAM memory?

AlquiWKT, 2022-05-15:
(spanish)
El Sketch usa 19064 bytes (59%) del espacio de almacenamiento de programa. El máximo es 32256 bytes.
Las variables Globales usan 1055 bytes (51%) de la memoria dinámica, dejando 993 bytes para las variables locales. El máximo es 2048 bytes.
(english)
Sketch uses 19064 bytes (59%) of program storage space. The maximum is 32256 bytes.
Global variables use 1055 bytes (51%) of dynamic memory, leaving 993 bytes for local variables. The maximum is 2048 bytes.

Doesn't seem like memory. I have tried commenting out most of the block which gives me problems and using Serial.println(scale.get_units(10)); , OLED display SH1106 doesn't work . other OLED SSD1306 works fine.
Thank you for your replay.

RobTillaart, 2022-05-16:
buenos días,

you might check with this function if your RUNTIME RAM is exceeded 
(works only on UNO and maybe on other AVR processors)

```cpp
uint32_t freeRam() 
{
    extern int __heap_start, *__brkval; 
    int v; 
    return (uint32_t) &v - (__brkval == 0 ? (uint32_t) &__heap_start : (uint32_t) __brkval); 
};
```

just place a call to ```Serial.println(freeRam());```  to see the runtime level of memory. 

look at the call stack of this call you mention
```cpp
Serial.println(scale.get_units(10));
```

get_units(uint8_t times)   calls   get_value(times)
get_value(times)  calls   a read  function depending on the **OPERATING MODE** 

the two median modes ( HX711_MEDIAN_MODE and HX711_MEDAVG_MODE) use more (runtime) memory than the HX711_AVERAGE_MODE and   HX711_RUNAVG_MODE. Never measured the difference, however I see an array of 15 floats allocated in both, + a call to insertSort() so it can easily consume 100+ bytes extra.

Which operating mode are you using?

AlquiWKT, 2022-05-16:
Buenos días, :)
I don't know the mode of operation. It is the one configured by default. I installed your libraries and tried the Hx_persistent example. This example worked fine for me. It only fails when I try to display something on the SH1106 OLED screen. If I use the SSD1306, I have no problems. Maybe the problem is in the SH1106 library.
I will try your suggestions to check the ram at runtime as soon as I have more time.
I've attached a zip with the example variation from your libraries in case it helps.

RobTillaart, 2022-05-16:
A discussion about the 1106 library tells me that the devices (1106 and 1306) are not 100% compatible
- https://forums.adafruit.com/viewtopic.php?f=25&t=51787

more info about differences
- https://picaxeforum.co.uk/threads/fyi-comparing-the-ssd1306-and-sh1106-oled-drivers.32387/
- https://forum.arduino.cc/t/whats-the-difference-between-adafruit-ssd1306-and-sh1106/247368/4

I am quite sure the problem is not in the HX711 library but in the display specifics.
Therefor I think you need to discuss this issue on the Arduino / Adafruit forum.

### RAM
You can save some runtime RAM by using the F() macro for constant text strings, e.g.

```cpp
  Serial.println( "Leer estructura desde EEPROM: " );
==> 
  Serial.println( F("Leer estructura desde EEPROM: ") );
```
This places the string in PROGMEM instead of in RAM.
Might help.

RobTillaart, 2022-05-16:
The default mode is set in reset()
```cpp
void HX711::reset()
{
  _offset   = 0;
  _scale    = 1;
  _gain     = 128;
  _lastRead = 0;
  _mode     = HX711_AVERAGE_MODE;
}
```
so that uses the least RAM. 

#### example
I checked your example and it compiles but I do not have a SH1106 display to verify it freezes runtime.

AlquiWKT, 2022-05-26:
closed, thank you. Maybe the problem is in the SH1106 library.

closed 2022-05-26 by AlquiWKT