RobTillaart/HX711 #23: Delay when trying to print to 7 segments

adas291, 2022-06-02:
I am trying to run this library on my scale project, with 7 segment display and a load cell. However interval of reading is crucial because i have to refresh display very quickly. When I use this library it creates some delay and there is a clear refresh rate drop even when i use scale.get_value(1) . Is there a way to make your library operate without any delay?

RobTillaart, 2022-06-02:
> Is there a way to make your library operate without any delay?

you might need to rewrite some parts if you really want to go for the performance limits.
But (there is always a but and in this case a big one) the device itself blocks for the most time. 

It is in the hx711.cpp line 56++

```cpp
float HX711::read()
{
  //  this BLOCKING wait takes most time...
  while (digitalRead(_dataPin) == HIGH) yield();
```
You  can verify this in the datasheet. 

----

If you want the max performance you should use **read()** directly and do only the math you need.
That would remove some function calls and other overhead like mode selection.
Don't know how much you would win. 

A possible faster option is to strip all the floating point math from **read()** and other functions.
Do the math in milligrams in **int32_t**. Warning: it might introduce rounding artefacts.

---- 

Your question tickles to introduce a new mode **HX711_RAW_MODE**
Created - https://github.com/RobTillaart/HX711/issues/24
Might be a solution for you too.

----

What is the refresh interval of the display? 
Note: humans can read only a few numbers per second, IMHO faster than 5x per second is overkill.

adas291, 2022-06-02:
I think i solved this delay shortcoming with the help of millis(). I use your function only one time every 100ms as at this interval there is no visible flicker and otherwise loop will go to refresh display and wont stop. However, raw mode would be useful too :)
```
if(millis()-now > 100){
      
      value = scale.get_units(1);
      now = millis();
    }
    //Value and refresh rate
    PrintNumber(value, 3);
```

closed 2022-06-02 by adas291