RobTillaart/HX711 #26: Value gives "nan"

gekke111, 2022-06-05:
Hi,
We are working on a school project. We are making a cocktailmachine.
We are using the code of the kitchen scale example.
If we use the example, it works perfectly.
But if we implement it in our own code it gives us the value "nan".
We are using a arduino nano v3.
Do you maybe know what the problem is?

Thank you,
Debroyer Sander & Niels Wellens
[GIP_barduino.zip](https://github.com/RobTillaart/HX711/files/8840186/GIP_barduino.zip)

RobTillaart, 2022-06-05:
Quick look but I don't see differences in the code except the clock/data pin of the HX711.
So I assume it is interference with one of the other devices connected. (too little information to make an educated guess)

What you could do to analyze is 
- comment everything except the HX711 code. 
- disconnect all hardware (except HX711)

This should work as it is identical (except pins) as the example.

Then
- add the LCD code + reconnect LCD
- check if it still works 

- add motors 
- check if it still works.

One thing that might happen is that the motors generate noise that disrupt the communication with the HX711. 
(which measures very tiny voltages).

You might improve the signal quality by using pull up resistors (2K2) on both the data line and the clock line of the HX711.

Other tips
- If you are using the Arduino IDE use CTRL-T regularly for readability
- add in the header the URL + version number of the libraries used.

Hopes this helps to get started.

RobTillaart, 2022-06-05:
update

I compiled your sketch - no HX711 connected -- and I get  NAN too, so I should double check the pins used.

RobTillaart, 2022-06-11:
As I see no more remarks I close the issue.
If the problem persists feel free to reopen it.

closed 2022-06-11 by RobTillaart