RobTillaart/HX711 #29: ESP8266

Rustie0125, 2023-01-25:
Hi Rob

Wondering if you are aware of the issue with ESP8266 and the HX711 
I tried your library and everyone else version all result in the same error, works great on AVR 

`05:24:31.575 ->  ets Jan  8 2013,rst cause:2, boot mode:(3,7)
05:24:31.575 -> 
05:24:31.575 -> load 0x4010f000, len 3424, room 16 
05:24:31.575 -> tail 0
05:24:31.575 -> chksum 0x2e
05:24:31.575 -> load 0x3fff20b8, len 40, room 8 
05:24:31.622 -> tail 0
05:24:31.622 -> chksum 0x2b
05:24:31.622 -> csum 0x2b
05:24:31.622 -> v00042970
05:24:31.622 -> ~ld
`

RobTillaart, 2023-01-25:
Thanks for reporting,
No I am not aware, I only used it on AVR, and tested compilation on other platforms. 
(edited your post for layout)

Do not have time (by choice of course) to test all my libraries on all platforms.

RobTillaart, 2023-01-25:
Which example does report this?

At what moment does it fail, boot time (assume)?

trying to understand first line.
based upon
- https://riktronics.wordpress.com/2017/10/02/esp8266-error-messages-and-exceptions-explained/

**rst cause:2** External reset using reset pin or wake up from deep sleep**
**boot mode :  (3,6)**  ->  GPIO 15 is low, GPIO 0 is high, GPIO 2 high, esp8266 booted from flash memory. SDIO HighSpeed v1 IO, UART1 booting.

RobTillaart, 2023-01-25:
Thanks for this update, appreciated,

If needed I will update the library with your results / insights.
It might be good to create a (minimalistic) example specific for the ESP8266 when all things work. (e.g. plotter example)

Rustie0125, 2023-01-25:
Will be great.

This is part of a IoT project for my YouTube channel so will link this back
to you. Once posted.

With regards to accuracy. And linear scalability. I'm using x4 50kg load
cell setup and I seem to be able to setup and calibrate the system to be
accurate in certain parts of the resistive range and not the other.

Example. If I calibrate for 100kg load. It's accurate around the 80 to
130kg region but 2kg way off. And vs versa. I'm considering building a
lookup table with atleast 3 calibration points.

Have you ran into this before ?

RobTillaart, 2023-01-25:
> Have you ran into this before ?

No, used not those "big ones", only setups with one sensor 5 Kg and smaller.

I know there is difference in accuracy with the smaller loadcells , price is often an indicator for quality. From my head 0.5 % is very good already. Furthermore in my projects I kept max load < 90% of the devices to prevent damage.
Datasheet might tell details you want to know.

Read once that the stability of the power supply can cause deviations, cannot confirm this btw. I did notice in one project there was a temperature effect, solved by calibrating the system at operating temperature which was well above room temperature.

(thinking out loud)
If I needed to calibrate 4 loadcells, I would check each one separately for its range stability. 
A single bad one can disturb a combined measurement => imagine module B is behaving non-linear between 30-50kg
If they all are linear within the range, then it should be valid to add them together. Note the deviations can still add up. So 4x 0.5% can (worst case) still become 2%.

Rustie0125, 2023-01-25:
Thanx for info Rob.

I suspect that approach works for very small measuring ranges, from what I
can't find, even the professional loadcells they all have a non linear
response over and above the temperature range.

So I'm thinking of a 3 point calibration look up table sketch..

But anyway. Yes manage to get your library working on ESP8622 it was just
the pin assignment.

You can decide to make a example I guess that has this option

Or point them to a pin map. This is the spruce of truth that managed to
help me solve it.

https://randomnerdtutorials.com/esp8266-load-cell-hx711/

Thanx for the help again

RobTillaart, 2023-01-26:
Multi point calibration can be done with multiMap, github.com/robtillaart/multimap.

It is written for non-linear mapping of sensors, signals etc. It uses 2 float arrays of equal length, input values and corresponding output values. It does linear interpolation between points. The input values do not need to be equidistant, but need to be in ascending order. Furthermore they must cover the whole working range. The non equidistant allows to minimize the number of points for a requested accuracy.

Maybe create an HX711-PRO class that uses multimap to convert raw measurements to weight in one step. Allows the user to define the number of calibration points and their value to meet his accuracy needs.

RobTillaart, 2023-01-29:
@Rustie0125 
Made any progress?

Rustie0125, 2023-01-29:
Hey Rob..

Still working on it. Not sure I'll get to creating a class, would be nice
however.

I'm stuck in getting the loadcells integrated mechanically at the moment

RobTillaart, 2023-01-29:
I might have some time to create a class.Already spending lots of time on libs so one more is not a big issue.

Think it can be made a derived class from the current with as extra calibration function to fill a multimap array. 

What would be a good name? HX711-PRO ?

Rustie0125, 2023-01-29:
That sounds , happy to feature it in video.

To give you more context. As a proof of outside the box thinking for IoT
Implementation I'm integrating 4 load cells into my shoes( 4 loadcells
each) plus height sensors.

The idea is to track my weight distribution  thru the house and elevation
which will allow me to see how I walk. How much food/drink I eat in a day.

Weight magenement

Can be used to see if someone forgot car keys or phone when left the house
with it and didnt come back with it.

It's complelty impractical but it makes a valid point when trying to
explain  the concept of IoT and connecting the world.

Now, I weigh 100kg (220lb) and to be accurate to the weight of food or
phone in my pocket. It needs to be very sensitive hence hunt for accuracy
across the entire range.

With regards to names

HX711- Advanced, Pro, Array Calibration something like that sounds good but
you are the create so anything sounds good 👍

RobTillaart, 2023-01-29:
Looking at the interface, I think only the **void calibrate_scale(uint16_t weight, uint8_t times = 10);** call needs rework.
instead of one weight, it should provide multiple points somehow.

> I'm stuck in getting the loadcells integrated mechanically at the moment

Hope it is solvable

Rustie0125, 2023-02-02:
Hey Rob

Are you able to help with this Library ?

Rustie0125, 2023-02-02:
> Already started some investigation, and it takes time.(and I have quite
> some work)
>
> Easiest way to get it to work is
>
>    - use the existing library.
>    - calibrate zero and max weight.
>
> Use the returned weight as input in the multmap function.
> This latter does raw weight in, corrected weight out.
>
> It is not performance optimized but will work.

Makes sense. I'm not rushing you, just wanted to see if I'm on my own here.

Thanx for heads up..

RobTillaart, 2023-02-02:
@Rustie0125 

I adjusted the plotter demo sketch to have a nine point non-linear correction/calibration with multimap.

```cpp
//
//    FILE: HX_plotter_multimap.ino
//  AUTHOR: Rob Tillaart
// PURPOSE: HX711 demo
//     URL: https://github.com/RobTillaart/HX711

#include "MultiMap.h"
// inkg[] holds the weights from the linear calibrated HX711 sensor
float inkg[]  = { 0, 6.25, 11.50, 14.50, 30.00, 34.10, 38.20, 44.50, 50.00};
// outkg[] holds the corrected weight
float outkg[] = { 0, 5, 10, 15, 30, 35, 40, 45, 50};

#include "HX711.h"
HX711 scale;
uint8_t dataPin = 6;
uint8_t clockPin = 7;

float raw, corrected;

void setup()
{
  Serial.begin(115200);
  // Serial.println(__FILE__);
  // Serial.print("LIBRARY VERSION: ");
  // Serial.println(HX711_LIB_VERSION);
  Serial.println();

  scale.begin(dataPin, clockPin);

  // TODO find a nice solution for this calibration..
  // load cell factor 20 KG
  // scale.set_scale(127.15);
  // load cell factor 5 KG
  scale.set_scale(420.0983);       // TODO you need to calibrate this yourself.
  // reset the scale to zero = 0
  scale.tare();

  //  dump whole range 0..50 kg
  for (float w = 0; w <= 50; w += 0.1)
  {
    Serial.print(w, 3);
    Serial.print('\t');
    corrected = multiMap<float>(w, inkg, outkg, 9);  //  9 is the size of the arrays!
    Serial.println(corrected, 3);
  }
  delay(60000);
}

void loop()
{
  raw = scale.get_units(5);
  corrected = multiMap<float>(raw, inkg, outkg, 9);  //  9 is the size of the arrays!

  Serial.print(raw, 3);
  Serial.print('\t');
  Serial.println(corrected, 3);

  delay(250);
}

// -- END OF FILE --
```

You might wrap the arrays and the MultiMap call in a separate function.

RobTillaart, 2023-02-02:
here the output from Serial plotter to show the output from the loop in setup

raw = blue, from the HX711
red = corrected, after multimap

[image]

RobTillaart, 2023-02-02:
You can adjust the number of array elements to 3 or 21 or whatever number. (more is slower)
Then add every weight and the should be value in the two arrays.

This should work for you. 

To have the "same flexibility build in" is not difficult, but it is hard to make it easy configureable.
And I want the API for the class to be as simple as possible.

RobTillaart, 2023-02-09:
@Rustie0125 
Have you tried the multimap approach?

----

I have given a new HX711 lib some thoughts to embed multiple calibration points but I do not have a easy to use interface yet.

#### Requirements
- it must allow to change the number of calibration points runtime
- it must allow to change the (individual) calibration points runtime
- it must have an easy to configure default behavior (2 points linear calibration I assume)

#### Con
This implies dynamic memory allocation which is not preferred in embedded.
Can allocate10 calibration points in front as a max to prevent dyn.mem.

#### Interface
- **bool calibrateMultiPoint(nrPoints, float * weights)** set the nr of calibration points (must be >= 2.)
  weights is an external array representing the OUT array of the multimap.
  weights has an element 0 which is 0kg (tare)
  weights has an element  max weight at the end. 
  weights is an array with increasing elements.
- **bool calibrate(weight)** weight is one of the elements of the weights array.
  the function searches the appropriate index in the OUT array and fills in the raw measurement in the internal IN array.
  returns false if not found.
- **bool calibrateClear()** removes all raw measurements (all raw to zero?)
- **bool calibrationOK()** checks if all weights are done (no zero elements) and that every element > previous 

It is the responsibility of the user to calibrate all correct weights including 0.

Rustie0125, 2023-02-11:
Hi Rob sorry for the delay. Work been hectic.

I'm still figuring out the mechanics of it but it seems like you have
figured out the multipoint calibration delay. See comments below.

   - it must allow to change the number of calibration points runtime.
   - it must allow to change the (individual) calibration points runtime

I think allowing the number of calibration points at pre compilation is not
end of the world. The type of user that will be looking for this won't be
alot and your general calibration would suite them. But here my thinking
pre compliation. Based on the number inputs you can select what parameters
are asked for in runtime. This will reduce or increase the steps only where
needed.

Another option is simply to to ask the user what range they are looking to
measure. Minimum and maximum and auto calculating how many points would be
needed to achieve a linear response.

   - it must have an easy to configure default behavior (2 points linear
   calibration I assume) agreed.

Con

This implies dynamic memory allocation which is not preferred in embedded.
Can allocate10 calibration points in front as a max to prevent dyn.mem.

Same comment applies here, I think the type of person looking for this
would hardly require the range that would require 10 points. But a way to
handle this is simply to base it on the range the user enters and deciding
how many points are needed to a maximum of 10. As an example. For me 10
points would mean every 10kg since I have 100kg range. I suspect 4 point
would be enough already ?

Thoughts

Name Johann or (Jay) by the way

RobTillaart, 2023-03-10:
@ahaverty 
No problem you dropping in. Good to hear about applications
My only problem is to find enough time as I am self employed and that has higher prio as bills have to be paid .

Above I proposed a working solution but did not got feedback on it. Technically I figured how it should work but still have second thoughts about the API. 
To be continued.

Rustie0125, 2023-03-10:
> Hi Rob sorry for the delay. Work been hectic.
> 
> I'm still figuring out the mechanics of it but it seems like you have
> figured out the multipoint calibration delay. See comments below.
> 
> 
>    - it must allow to change the number of calibration points runtime.
>    - it must allow to change the (individual) calibration points runtime
> 
> I think allowing the number of calibration points at pre compilation is not
> end of the world. The type of user that will be looking for this won't be
> alot and your general calibration would suite them. But here my thinking
> pre compliation. Based on the number inputs you can select what parameters
> are asked for in runtime. This will reduce or increase the steps only where
> needed.
> 
> Another option is simply to to ask the user what range they are looking to
> measure. Minimum and maximum and auto calculating how many points would be
> needed to achieve a linear response.
> 
>    - it must have an easy to configure default behavior (2 points linear
>    calibration I assume) agreed.
> 
> Con
> 
> This implies dynamic memory allocation which is not preferred in embedded.
> Can allocate10 calibration points in front as a max to prevent dyn.mem.
> 
> Same comment applies here, I think the type of person looking for this
> would hardly require the range that would require 10 points. But a way to
> handle this is simply to base it on the range the user enters and deciding
> how many points are needed to a maximum of 10. As an example. For me 10
> points would mean every 10kg since I have 100kg range. I suspect 4 point
> would be enough already ?
> 
> Thoughts
> 
> Name Johann or (Jay) by the way
> 
> 
> 
> > @Rustie0125 <https://github.com/Rustie0125>
> > Have you tried the multimap approach?
> > ------------------------------
> >
> > I have given a new HX711 lib some thoughts to embed multiple calibration
> > points but I do not have a easy to use interface yet.
> > Requirements
> >
> >    - it must allow to change the number of calibration points runtime
> >    - it must allow to change the (individual) calibration points runtime
> >    - it must have an easy to configure default behavior (2 points linear
> >    calibration I assume)
> >
> > Con
> >
> > This implies dynamic memory allocation which is not preferred in embedded.
> > Can allocate10 calibration points in front as a max to prevent dyn.mem.
> > Interface
> >
> >    - *bool calibrateMultiPoint(nrPoints, float * weights)* set the nr of
> >    calibration points (must be >= 2.)
> >    weights is an external array representing the OUT array of the
> >    multimap.
> >    weights has an element 0 which is 0kg (tare)
> >    weights has an element max weight at the end.
> >    weights is an array with increasing elements.
> >    - *bool calibrate(weight)* weight is one of the elements of the
> >    weights array.
> >    the function searches the appropriate index in the OUT array and fills
> >    in the raw measurement in the internal IN array.
> >    returns false if not found.
> >    - *bool calibrateClear()* removes all raw measurements (all raw to
> >    zero?)
> >    - *bool calibrationOK()* checks if all weights are done (no zero
> >    elements) and that every element > previous
> >
> > It is the responsibility of the user to calibrate all correct weights
> > including 0.

@RobTillaart  how is this not feedback ? What are you expecting ?.

RobTillaart, 2023-03-10:
Goodmorning Jay,

You have given a lot of feedback Ion the API which is appreciated.

What I meant was I do not recall that you have tried the example code that uses the current HX lib plus a call to multimap. That would be confirmation that the solution ideas are workable.
Sorry for being not explicit on this. I will reread this whole thread later to pick up pieces I missed.

RobTillaart, 2023-03-10:
- Created develop branch
- copied HX711 repo +rename to HX711_MP
- removed all examples minus plotter example (easier to add later than to keep them all in sync during develop)
- updated all meta files for build etc.
- checked all build-CI files
- Created a PR to have a starting point.

RobTillaart, 2023-03-10:
@Rustie0125 

[image]

first rough test, blue line is the raw data, the red line is 5 pieces of interpolated grams
The data is just dummy as I have no test setup (other projects have prio for room)

Code snippet
```cpp
  //  dummy data, adjust to your scale.
  scale.setCalibrate(0, 1000, 0);
  scale.setCalibrate(1, 1300, 10000);
  scale.setCalibrate(2, 2000, 20000);
  scale.setCalibrate(3, 4000, 30000);
  scale.setCalibrate(4, 5000, 40000);
  scale.setCalibrate(5, 5500, 50000);

  for (uint32_t raw = 1000; raw <= 7000; raw += 20)
  {
    Serial.print(raw);
    Serial.print("\t");
    Serial.println(scale.testCalibration(raw));
  }
```

- **scale.setCalibrate(index, raw, weight)** so at index 0, we have a raw data of 1000 and corresponding weight of 0
- **scale.testCalibration(raw)** is a debug function to see if raw values give the right weight.

The idea is that you calibrate the sensor in setup() and thereafter you just call **get_value()** or **get_unit()**

The numbers for the calibration can be adjusted runtime, but the index + raw must be an increasing array.

RobTillaart, 2023-03-10:
@Rustie0125 

- Cleaned up the code and got example compiling again.
- updated readme.md, keywords etc.
- Made this plot (again with dummy data)
  - it has a range that starts at -10 Kg - 90 Kg
  - 10 data points  (0..9)

Code will be in develop branch in a few minutes.

[image]

Rustie0125, 2023-03-10:
Rob this is excellent 👏

I'll make sure to test it this weekend. Timing perfect aswell received
additional load amplifier and cells yesterday so will try setup and run the
two system independently and see how the accuracy scales and how much
deviance there between two different setups with the same calibration
parameters

RobTillaart, 2023-03-10:
@Rustie0125, @ahaverty 

Took quite some time as I had to stabilize HX711 first before I could split of the code. Not a big problem but all in all it consumes quite some time in effect most part of the day.

It was not clear what to do with the **tare()** function and the related **offset** functions, in the end I decided to keep it as simpel as possible. **tare()** first became a wrapper function around **setCalibrate()** but soon it "left the building" altogether as it made the API in fact a bit ambiguous and therefor more complex. 

The consequence is that the **get_value()** function et al differs from the HX711 library. This means they are not one to one interchangeable (e.g. HX711_MP with 2 points behaves differently). This is more true as the zero point does not need to be the index = 0 of the array. ( due to the support of non linear negative forces ).

I have to make notes of this discrepancy in the readme file of both libraries. 

If there are issues with the multi point version, please open separate issues under that repo.
If they affect also the HX711 repo please mention it.

RobTillaart, 2023-03-11:
@Rustie0125, @ahaverty

- added performance examples (from HX711 lib)
- added a minimal runtime calibration example
- added **adjustCalibrationRaw(index, amount)** for easier adjustments
- updated readme.md (again)
- minor edits

RobTillaart, 2023-03-11:
@Rustie0125, @ahaverty

- did some final micro edits, indicating it is time to stop :)
- will merge into master if build succeeds and release version 0.1.0 today (so there will be more feedback)

Please open issues if there are problems,

Rob

RobTillaart, 2023-03-11:
Released as 0.1.0

RobTillaart, 2023-03-11:
I close this issue here as it is not an issue for this library anymore.

closed 2023-03-11 by RobTillaart