RobTillaart/HX711 #36: Display Peak Value

jclark0193, 2023-05-28:
I have a meter that needs to display only the peak value sensed by the load cell.

RobTillaart, 2023-05-29:
Thanks for the question.
Just add a global variable to your project.
Initialize it with the first read of the load cell.
Compare every next measurement with this var and keep the largest.
Display the var when needed.

(Note this is not a function that will be build into the library)

jclark0193, 2023-05-29:
Thanks Rob. I just purchased my first Arduino, with zero experience programming. Trying to build my first project. Thanks for the help.

RobTillaart, 2023-05-29:
Start small, look at other peoples sketches and try to understand how they work.

RobTillaart, 2023-06-12:
As issue has no concrete question anymore it is closed.

closed 2023-06-12 by RobTillaart