RobTillaart/HX711 #39: Tare() function not working as per intended.

atocec, 2023-07-04:
I have calibrated my load cell with weight 1100 gram and got ,
Calibration factor value   : -21
Offset value  :  1559

then i have stored these values in EEPROM . After that whenever i call Scale.tare() function values should start counting from 0, but it sometime start from positive random numbers or Negative numbers.

so can u tell me that is there anything i missing or this is the problem from library side.

RobTillaart, 2023-07-04:
Thanks for filing this issue.

To find the cause of the problem I need to analyze the code you use.
**Q: Can you post the code you use?**

----

> Calibration factor value : -21
> Offset value : 1559

These value differ per loadcell used so they are specific. The factor of 21 looks reasonable compared with my test load cell, the offset 1559 is nowhere near my test load cell.

----

**Scale.tare()** is best called with a parameter 20 or so **Scale.tare(20)** so it takes the average of 20 measurements. 
As the HX711 has quite some noise it does in practice never get a perfect 0 weight. Multiple readings are needed.
If your setup has extra mechanical noise, even more measurements may be needed.

----

> let me give one example:-
→ when i applied the weight of 6 kg on it and send command of tare() then values start counting from aroud 40 then again i send the tare command it then start counting from 60 , next time some negative value .

using a fixed weight to set a "virtual" zero point should be possible if the range of the loadcell is large enough.

**Q: What is the range of the loadcell used?**

RobTillaart, 2023-07-04:
> or this is the problem from library side.

The library has been around for 3++ years, and I do not recall such issue mentioned before.

If accuracy (non-linearity) over the range is an issue the **HX711_MP** library allows multiple calibration points.
However harder to configure.

RobTillaart, 2023-07-12:
@atocec 
Is this still an issue?

atocec, 2023-07-13:
@ RobTillaart/HX711

No , there was problem from our code side.

    When you ask for code then I just make one test code for you and I
found that test code working properly .
 so i started debugging my original code by comparing test code i
found that problem.

    Thank you for your concern and pardon me for the delay in reply.

bbbbmmdddd, 2024-08-08:
您好，我也遇到了这个问题，请问呢您是怎么解决的呢？ @ATOC-39