RobTillaart/HX711 #40: Multiple HX711 sharing CLK?

SanderM2, 2023-08-18:
Is it possible to have multiple HX711 connected and have them all shared the same CLK pin?
I'm running out of pins on my arduino and sharing the CLK should be possible I think? Just not sure if this library supports it.

RobTillaart, 2023-08-18:
Need to think about it and it is 10:30 pm here so I will come back to this tomorrow.
First thoughts are that it is safer to share data lines.
If the hx711 gets clock pulses it will act somehow 
Good question, needs some datasheet checking.

SanderM2, 2023-08-18:
I thought sharing the clock line was the right way to do it. The MCU can ignore data on the other data lines when it's not actively reading that one. So it shouldn't matter if other HX711 are responding, right?

Sharing data lines would mean we're going to connect all DOUT pins of each HX711 together? It's an output only pin so I don't think that's a good idea?

Would love to hear from you tomorrow. Have a good night!

RobTillaart, 2023-08-18:
Oops you're right 
You see it is definitely late :)

RobTillaart, 2023-08-19:
@SanderM2 

Read the data-sheet after a good sleep :)

----

**WARNING: Sharing the data lines is NOT possible as it could cause short circuit.**

----

Sharing the **CLOCK** lines will work and as said before it will have some side effects.
If these are acceptable depends on the project at hand and the setup.

The side effects I found so far: (page 4 and 5 datasheet)
- The **CLOCK** is used to select channel and to select gain for the NEXT sample. 
- The **CLOCK** is used for power down. 
- After wake up after power down all HX711's will reset to channel A and gain 128.

So in short, sharing the **CLOCK** line causes all HX711 modules share the same state.
This can introduce extra complexity if one uses mixed gains or channels. 
If all HX711's use the same settings it should work.

----

Alternative is to use a multiplexer e.g. [HC4052](https://github.com/RobTillaart/HC4052)

RobTillaart, 2023-08-19:
> The MCU can ignore data on the other data lines when it's not actively reading that one

Correct, if you create two HX711 objects with the library, only the object that calls **read()** will fetch data.

> So it shouldn't matter if other HX711 are responding, right?

See the above side effects.

RobTillaart, 2023-08-19:
FYI, I prepared some notes to add the above findings to the readme.md file.

RobTillaart, 2023-08-22:
Another idea popped up, just to share.

If you add AND gates (74LS08) to the HX711's you can create multiple SELECT lines,
selecting only one HX711 at a time. 
This can be extended many times.

```
      MCU                         AND GATE                 HX711
   +---------+                   +---------+             +---------+
   |     DIN |<--------+---------| OUT   A |<------------| OUT     |
   |         |         |         |         |             |         |
   |    SEL1 |==================>| B       |             |         |
   |         |         |         +---------+             |         |
   |     CLK |----+------------------------------------->| CLK     |
   |         |    |    |                                 +---------+
   |         |    |    |
   |         |    |    |
   |         |    |    |          AND GATE                 HX711
   |         |    |    |         +---------+             +---------+
   |         |    |    +---------| OUT   A |<------------| OUT     |
   |         |    |              |         |             |         |
   |    SEL2 |==================>| B       |             |         |
   |         |    |              +---------+             |         |
   |         |    +------------------------------------->| CLK     |
   |         |                                           +---------+
   +---------+ 
```

RobTillaart, 2023-08-23:
Added example **HX_loadcell_array.ino** in develop branch.
Tested this (< 1 hour) with hardware and it worked without problems.

Will be integrated in master with an upcoming release (probably 0.3.8), timeline unknown.

SanderM2, 2023-08-24:
Thanks! ;-)

I've been experimenting with your library earlier today. With a single HX711 for testing now it seems like get_units(10) takes approximately 800ms. Isn't the HX711 supposed to be able to go faster?

RobTillaart, 2023-08-24:
@SanderM2 
> With a single HX711 for testing now it seems like get_units(10) takes approximately 800ms. 

The sensor blocks between reads, and it makes typically 10SPS (see below)
```cpp
hx711.cpp about line 90

//  From datasheet page 4
//  When output data is not ready for retrieval,
//       digital output pin DOUT is HIGH.
float HX711::read()
{
  //  this BLOCKING wait takes most time...
  while (digitalRead(_dataPin) == HIGH) yield();
```

Can you run the **HX_performance.ino** sketch and post your output?

----

> Isn't the HX711 supposed to be able to go faster?

The HX711 can be configured to 10 SPS or 80 SPS, however as all breakouts I have used 
do not expose the **RATE** pin I have no way to let it go faster,
See - https://github.com/RobTillaart/HX711/tree/develop#10-or-80-sps

If there is an explicit need I have planned future support on the **could** list
- https://github.com/RobTillaart/HX711/tree/develop#could

RobTillaart, 2023-08-24:
@SanderM2 

If the blocking is a problem for your project you could 
- do the averaging yourself and make a single measurement every ~100 milliseconds.
- use https://github.com/RobTillaart/RunningAverage
- use https://github.com/RobTillaart/RunningMedian
- use https://github.com/ripred/Smooth

RobTillaart, 2023-08-26:
@SanderM2 
As the sharing CLK seems to work, I propose to close this issue. 
Feel free to open an issue on the performance and support of the **RATE** pin in the library.

RobTillaart, 2023-09-03:
As there is no additional questions/remarks I close the issue.
Feel free to reopen if needed.

closed 2023-09-03 by RobTillaart

wmurw, 2025-06-06:
> Another idea popped up, just to share.
> If you add AND gates (74LS08) to the HX711's you can create multiple SELECT lines, selecting only one HX711 at a time. This can be extended many times.

Only this idea kills the idea of reducing Arduino pins.