RobTillaart/HX711 #43: Multiple HX711 using MCP23017

middlegrey, 2023-10-07:
Firstly, thanks for all your work!

Is it possible to connect multiple HX711 chips to an Arduino using MCP23017 port expanders?

RobTillaart, 2023-10-07:
Never tried.
Technically I think it would be possible however the library does definitely not support it. The library would need drive the clock and data line with mcp23017 calls. So the lowest level of the library needs to be replaced.

RobTillaart, 2023-10-07:
Have a look at the HC4052 multiplexer. It would allow one HX711 object talk to 4 devices. Just set the channel to 0..3 to select device. 
Less complex for sure.

middlegrey, 2023-10-09:
Thanks for the quick response.

I've tested the MCP23017, HC4051 and HC4052 but can't seem to get any to work. The HX711 seems to calibrate in the setup but then just prints the same (incorrect) value in the loop. With the MCP23017 the value changes if I add weight at the exact moment it performs a read, but it's not changing constantly like when I connect the HX711 directly to the Arduino.

I'll keep trying when I have some more time.

RobTillaart, 2023-10-09:
Mmm, of course (bright moment of insight)
Every HX711 has its own calibration / tare so when changing channel these should be adjusted too.

Alternative(?)
Use a co processor? Using a 2nd UNO to connect a number of HX devices and provide their values over Serial?
```
Id, weight
0, 0
1, 123
2, 0
3, 509
Etc
```
The receiving end can request a specific number by sending the id to the co-processor.
If the co-processor caches the last vslues, it can respond within milliseconds.

An UNO should be able to handle at keast 5 Hx711 devices. Using the analog pins for io could make that probably 9

middlegrey, 2023-10-09:
I'm needing to read 32 load cells so will go with an Arduino Giga to simplify things.

RobTillaart, 2023-10-09:
Not tested my library with the GIGA board but I see no reason why it should not work.
I'm interested if the library works well with the GIGA so please let me know.

RobTillaart, 2023-10-24:
Any progress with the GIGA board?

middlegrey, 2023-10-24:
Still finalising the hardware, hopefully next week

RobTillaart, 2023-11-21:
Any progress with the GIGA board?
Can the issue be closed?

closed 2023-11-24 by RobTillaart