RobTillaart/HX711 #45: Request for Power Saving Feature

cbijon, 2023-12-17:
As a user of the HX711 library, I have noticed that there is no built-in power saving feature to reduce power consumption during idle periods.

Request:
I would like to propose the addition of a power saving feature in the HX711 library. This feature could include methods like .powerUp() and .powerDown() to manage the power state of the HX711 module. This would be particularly useful for scenarios where power consumption needs to be minimized during periods of inactivity.

Use Case:
Provide an example or scenario where having a power saving feature would be beneficial, such as when the HX711 is used in a battery-powered device where energy efficiency is crucial.

Potential Implementation:
If you have any suggestions on how this feature could be implemented, please share your ideas. If not, express your willingness to collaborate on finding a suitable implementation.

Additional Context:
Include any additional information that might be relevant to the request, such as any existing workarounds or alternative solutions that you've explored.

RobTillaart, 2023-12-17:
Thanks for the request, will take several days before I have time to dive into it as I have a lot of work in progress.

RobTillaart, 2023-12-17:
Some first thoughts… four functions needed:
- **void setPowerPin(uint8_t pin)** make it possible.
- **bool powerUp()** idem
- **bool powerDown()** idem
- **bool isPowerOn()** check state.

Possibly one need a timestamp when powered up.
If no power pin is defined the up/down functions will return false.
Default state must be "POWERUP", to be backwards compatible and to be usable for people who do not use this feature.

Possibly a derived class implements these extra functions.
If the user wants to use the feature it is his responsibility to check state before reading a load.

The feature does not implement battery monitoring or voltage level or current current. Outside scope of a loadcell.

closed 2023-12-18 by RobTillaart