RobTillaart/HX711 #46: Fast processor read() delay

palmerr23, 2024-03-02:
Thanks for the library, it works well!

To answer the question embedded in the code at line 135 in the cpp file, you do not need the delayMicroseconds(1) for a Pico running at 240MHz (PD_SCK is 500nS) in _shiftIn( ) **or** read( ).  

The delay( ) treatment should probably be the same in both places.

It would be useful, however to have an option, as a Teensy 4 and other processors running at > 500MHz are likely to need them in **both** read( ) **and** _shiftIn( ).
```
  while (m > 0)
  {
    //  delayMicroSeconds(1) needed for fast processors?
    digitalWrite(_clockPin, HIGH);
    digitalWrite(_clockPin, LOW);
    m--;
  }
```

I hope this is useful.

RobTillaart, 2024-03-02:
Thanks for this feedback,
Definitely useful to hear.

RobTillaart, 2024-03-02:
- Merged the PR,
- Will increase version asap
- will add the change also to HX711_MP repo (asap)

RobTillaart, 2024-03-03:
@palmerr23 
HX711 & HX711_MP released
Again thanks for your addition / feedback

closed 2024-03-03 by RobTillaart