RobTillaart/HX711 #49: Platform IO considers the changes from 0.3.1 to 0.4 breaking and refuses smooth updates

pstimpel, 2024-05-21:
Hi,

I have not idea if this is an issue of this project, or what has to be done to make platform.io not considering regular updates "breaking". But I thought I would bring this to your attention. And no, I have absolutely no idea what has to be done about it. 

Cheers,
Peter

RobTillaart, 2024-05-21:
Hi Peter,

Have no clue either, will look into this coming days (busy for another project)
It could affect all my libraries, so it has some prio.

Thanks for reporting
Rob

RobTillaart, 2024-05-21:
@ivankravets
Is there any problem at platformIO related to this issue?

@pstimpel 
Can you elaborate what you see when you update, or did not see what you expected?

pstimpel, 2024-05-21:
There is a feature to update liraries of your projects. if there are updates, one can just press a button and the project receives an update for that lib, downloads the stuff from the repo, and that's it. Quite convenient ... 

However, sometimes such updates get marked "incompatible", and usually that means that function headers changed and the Arduino  project becomes incompatible with the new library version. In this case one should review the code once he/she raised the version of the depending libs. This is quite a useful measure to keep the projects in shape. 

But the update from 0.3.1 to 0.4.0 did not contain such changes as  far as I can tell, which makes me thinking maintainers have to declare compatibility somehow. I do not know how, or where such a declaration would happen. But it seems to me, your project could benefit if you can make it "compatible". 

If you decide not to do so, there is nothing crazy happening, it is just a manual step needed for updating to a newer version, instead of that once-click thing. 

And no, I found no issues because of this "missing declaration", it is for me just a convenience thing. 

BR,
Peter

RobTillaart, 2024-05-21:
> which makes me thinking maintainers have to declare compatibility somehow

version x.y.z should be an indication.
It could be that the version number 0.3.9 => 0.4.0 sets this incompatible flag, that is simplest way to set the flag.

Note that the **begin()** function got an extra parameter **fastProcessor**, 
so there is at least one change of interface which can be detected ==> set the flag!.

```cpp
void begin(uint8_t dataPin, uint8_t clockPin, bool fastProcessor = false);
```

ivankravets, 2024-05-21:
@RobTillaart , there is no problem with PlatformIO. What the user asks is to maintain the [SemVer](https://semver.org/) philosophy when incrementing the library version. If you have breaking changes, you have to increment the MAJOR number, if you have new features but keep compatibility with API from the previous version, you increment the MINOR number. Finally, if you fixed any issues, you increment the PATCH number.

@pstimpel , you can solve this issue on your side using [lib_deps](https://docs.platformio.org/en/latest/projectconf/sections/env/options/library/lib_deps.html) and [Package Specification](https://docs.platformio.org/en/latest/core/userguide/pkg/cmd_install.html#cmd-pkg-install-specifications). Also, see https://docs.platformio.org/en/latest/core/userguide/pkg/cmd_install.html#cmd-pkg-install-requirements

So, the final solution:
```ini
[env:myenv]
platform = ...
lib_deps = 
     robtillaart/HX711@~0.4.0
```

pstimpel, 2024-05-21:
Thanks for the quite good explanation, @ivankravets 

@RobTillaart , this explains why I did not find any issues with the changed version, it was an optional parameter. 

I am sorry if this was creating some headaches to you.

pstimpel, 2024-05-21:
@ivankravets , I am not getting what to change. I had this

> lib_deps = 
>      robtillaart/HX711@^0.3.1
> ```

when 0.4.0 was released. So, it was a raise in the minor version number, aka "no API changes". In my understanding, platform.io should allow the update directly, no? In my opinion, @RobTillaart did not create an incompatible API, so he was right in changing from 0.3.1 to 0.4.0

Do I understand you and the docs wrong?

ivankravets, 2024-05-22:
You need to use `~` (allow only bug fixes) instead of `^`.

RobTillaart, 2024-05-22:
> ... I am sorry if this was creating some headaches to you.

No problem, it is good to refresh ones mind regularly. 

If it solves your question, you may close the issue.

RobTillaart, 2024-05-22:
@ivankravets Thanks for your input, appreciated!

pstimpel, 2024-05-22:
@ivankravets 

> You need to use `~` (allow only bug fixes) instead of `^`.

But isnt this the opposite of what I was expecting?

From the docs: 

> ^1.2.3 - any compatible version (new functionality in a backwards compatible manner and patches are allowed, 1.x.x). RECOMMENDED
> ~1.2.3 - any version with the same major and minor versions, and an equal or greater patch version

My goal is to allow updates from 0.3.1 to 0.4.0. If patch or minor change (no API changes), allow the smooth one click update. So, ^0.3.1 should exactly do this, agreed?

Sorry for not stopping the discussion, but it is truly not clear to me how your hint for using ~ instead of ^ could solve the issue I had.

RobTillaart, 2024-06-15:
@pstimpel 
HI Peter,
Please close this issue as the discussion not active anymore.

pstimpel, 2024-06-15:
closing on your request, Rob. Not sure this was solved. 

Cheers, 
Peter

closed 2024-06-15 by pstimpel

RobTillaart, 2024-06-15:
@pstimpel 
Agree not solved, however not an issue of this library. I don't mind if an discussion is active but this one is not.

> My goal is to allow updates from 0.3.1 to 0.4.0. If patch or minor change (no API changes), allow the smooth one click update. So, ^0.3.1 should exactly do this, agreed?

As far as I understand the notation (theory from the docs)

~0.3.1 would allow only 0.3.Z version where Z = 1 and up, so it would NOT allow 0.4.0

^0.3.1 would allow 0.Y.Z version where (Y = 3 and Z = 1 and up)  or  (Y > 3 and Z = 0 and up)  
            ==> so it would allow 0.4.0, 0.4.1 etc

So if  your **from 0.3.1 to 0.4.0**  means **from 0.3.1 to and including 0.4.0**  you should use ^

pstimpel, 2024-06-15:
Yes, but as said before, I have @^0.3.1. So I would have expected the update to 0.4 not being incompatible. And this is where I think @ivankravets answer missed the point - or I am missing something completely.

And sure, nothing out of your library.