RobTillaart/HX711 #52: Loadcell array with multiple clock pins?

Ollie-Boyd, 2024-09-12:
Hi Rob, 
Great library and thanks for your work!
Is it possible to use your array example with multiple clock lines? For example I was trying something like this:
```
#include "HX711.h"

HX711 scale0;
HX711 scale1;
HX711 scale2;
HX711 scale3;

HX711 scales[4] = { scale0, scale1, scale2, scale3 };

const uint8_t dataPin[4] = { 12, 10, 8, 6 };
const uint8_t clockPin [4] = { 11, 9, 7, 5 };
```

But `clockPin` only accepts one element. 

Do you think this is possible? Thanks so much!

RobTillaart, 2024-09-12:
Thanks,
Should work as all the scales will have their own data and clock pin. 
Can you confirm?
(issue was closed before I could answer :)

Ollie-Boyd, 2024-09-12:
Hi Rob, yes thanks it's perfect