RobTillaart/HX711 #53: How does one get the raw data scaled in milivolts?

busypete, 2024-09-12:
I am using the HX711 on an experiment with a strain gauge, and do not need to convert or scale to weight units, I only need the output in +-20 milivolts (128 gain). However, I am struggling a bit around the concept of the conversion from the raw serial data from the ADC to the IEEE754 float mantissa. I cannot seem to match the data bits seen on the scope to the ones saved on the float. How can I extract or scale this data to represent the input in volts?

RobTillaart, 2024-09-13:
Interesting question, I need to do some reading.

RobTillaart, 2024-09-13:
>  I only need the output in +-20 milivolts (128 gain).

#### Process how I would try to get mV's (not tested)

Put 0.0 mV on the A port,
Call **read()** to get the zero point.  (Expect the ADC returns zero +- some noise)
Put 20.0 mV on the A port
Call read to get the max value. 
Put -20.0 mV on the A port
Call read to get the min value. 

Then interpolate between those points to get your measurements in milliVolts.
If needed more interpolation points can be needed (check HX711_MP)

If 0.0 mV equals a read of 12345
And max equals  24365155

In code you need something like
```cpp
float getmV()
{
  float zeroV = 12345.0
  float maxV = 24365155.0;
  float raw = HX.read();    <<<<<<<<<<<<<<<<<<< you might use different modi like average or median etc.
  return 20.0 * (raw - zeroV) / (maxV - zeroV) ;
}
```

Can you do the above test, I have no HW setup free to do so.
Please post your results.

If it works well I can add it as part of the interface of the HX711 class.

RobTillaart, 2024-09-13:
## Idea!

In a bright(?) moment I had the following thought:

- Call **tare()** when 0.0mV is applied 
- Call **calibrate_scale(20000)** when 20.000 mV is applied.

then the **float get_units(uint8_t times = 1);** should return voltages

RobTillaart, 2024-09-20:
@busypete 
Did the above replies answer your question?

RobTillaart, 2024-09-25:
@busypete 
As there is no answer to my question I assume the issue is solved. 
Feel free to reopen if this is not the case.

closed 2024-09-25 by RobTillaart

RobTillaart, 2024-09-25:
Added a section in the readme.md of my local develop branch to keep this idea.