RobTillaart/HX711 #58: Continuous Restart

sanluca, 2024-12-19:
Hi,

I'm having a problem with my ESP8266 since I updated the libraries to the latest version. I'm getting the following errors, and I believe it's related to this library. I'm attaching the error messages and the code I'm using. Thanks in advance for any help.

`08:43:47.957 -> Adjusted time 24146280. Offset = 101
08:43:50.530 -> 
08:43:50.530 -> User exception (panic/abort/assert)
08:43:50.561 -> --------------- CUT HERE FOR EXCEPTION DECODER ---------------
08:43:50.561 -> 
08:43:50.561 -> Panic core_esp8266_main.cpp:191 __yield
08:43:50.561 -> 
08:43:50.561 -> >>>stack>>>
08:43:50.561 -> 
08:43:50.561 -> ctx: sys
08:43:50.561 -> sp: 3fffff70 end: 3fffffb0 offset: 0010
08:43:50.561 -> 3fffff80:  0170017f 00000000 4021793a 4021795d  
08:43:50.561 -> 3fffff90:  3ffefe70 00000000 00000000 4021101b  
08:43:50.561 -> 3fffffa0:  3fffdad0 00000000 3fff03b8 3fff03e4  
08:43:50.561 -> <<<stack<<<
08:43:50.561 -> 
08:43:50.561 -> --------------- CUT HERE FOR EXCEPTION DECODER ---------------
08:43:50.594 -> 
08:43:50.594 ->  ets Jan  8 2013,rst cause:2, boot mode:(3,7)
08:43:50.594 -> 
08:43:50.594 -> load 0x4010f000, len 3424, room 16 
08:43:50.594 -> tail 0
08:43:50.594 -> chksum 0x2e
08:43:50.594 -> load 0x3fff20b8, len 40, room 8 
08:43:50.594 -> tail 0
08:43:50.594 -> chksum 0x2b
08:43:50.594 -> csum 0x2b
08:43:50.594 -> v0005d7c0
08:43:50.594 -> ~ld
08:43:50.683 -> ����'�{��g|�d�l`c��<{�l�o��o�l`��r�l�$��
08:43:52.038 -> setLogLevel: ERROR | STARTUP | DEBUG |
08:43:52.082 -> STARTUP: init(): 0
08:43:52.161 -> STARTUP: AP tcp server established on port 5555
08:43:53.059 -> DEBUG: MD5 found cbab056738488885e22eff967c78abc9
08:44:19.732 -> 
08:44:19.732 -> User exception (panic/abort/assert)
08:44:19.732 -> --------------- CUT HERE FOR EXCEPTION DECODER ---------------
08:44:19.732 -> 
08:44:19.732 -> Panic core_esp8266_main.cpp:191 __yield

Decoding stack results
0x4021793a: __preloop_update_frequency() at C:\Users\sanlu\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.2\cores\esp8266\core_esp8266_main.cpp:99
0x4021795d: can_yield() at C:\Users\sanlu\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.2\cores\esp8266\core_esp8266_main.cpp:116
0x4021101b: std::function<bool at )>::operator=(std::function<bool ()> const&) (c:\users\sanlu\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\3.1.0-gcc10.3-e5f9fec\xtensa-lx106-elf\include\c++\10.3.0\bits\std_function.h:303`

code arduino

`//************************************************************
// lettura presenza nel letto con HX711
// wemos d1 mini 
// painlessmesh 1.5.4
// esp8266 3.1.2
// 
//************************************************************
#include <HX711.h>
// HX711 Pins
const int LOADCELL_DOUT_PIN = 2; // Remember these are ESP GPIO pins, they are not the physical pins on the board.
const int LOADCELL_SCK_PIN = 3;
int calibration_factor = 400; // Defines calibration factor we'll use for calibrating. se diminuisce aumenta il valore inversamente proporzionale
float reading; // Float for reading
float raw; // Float for raw value which can be useful
unsigned long now;

HX711 scale;                          // Initiate HX711 library

//#include "painlessMesh.h"
#include "namedMesh.h"

#define   MESH_PREFIX     "whateverYouLike"
#define   MESH_PASSWORD   "somethingSneaky"
#define   MESH_PORT       5555

Scheduler userScheduler; // to control your personal task
//painlessMesh  mesh;
namedMesh  mesh;
String msg,rel;
String nodeName = "lettocamera"; // Name needs to be unique
String to = "bridgemqtt";
uint32_t root_id;

#define ROLE    "lettocamera"
#define VERSION "LettoCamera v2.0.1"
#define MESSAGE "lettocamera "

// User stub
void sendMessage() ; // Prototype so PlatformIO doesn't complain
void sendMessage1();

Task taskSendMessage( TASK_SECOND * 1 , TASK_FOREVER, &sendMessage );
Task taskSendMessage1( TASK_SECOND * 1 , TASK_FOREVER, &sendMessage1 );

void sendMessage() {
  
  read_peso();
  
  taskSendMessage.setInterval( random( TASK_SECOND * 300, TASK_SECOND * 400 ));
}

void sendMessage1() {
  update_status();
  taskSendMessage1.setInterval( random( TASK_SECOND * 600, TASK_SECOND * 700 ));
}

// Needed for painless library
void receivedCallback( uint32_t from, String &msg ) {
  Serial.printf("startHere: Received from %u msg=%s\n", from, msg.c_str());
  root_id=from;
  rel=msg;

    if (strcmp(rel.c_str(),"1") == 0){
      //tara la bilancia
      scale.power_up();
      scale.wait_ready();
      scale.set_scale();
      scale.tare();       //Reset scale to zero
      scale.power_down();
  } 
 }

void newConnectionCallback(uint32_t nodeId) {
    Serial.printf("--> startHere: New Connection, nodeId = %u\n", nodeId);
}

void changedConnectionCallback() {
  Serial.printf("Changed connections\n");
}

void nodeTimeAdjustedCallback(int32_t offset) {
    Serial.printf("Adjusted time %u. Offset = %d\n", mesh.getNodeTime(),offset);
}

void read_peso(){
  scale.power_up();
  scale.wait_ready(); // Wait till scale is ready, this is blocking if your hardware is not connected properly.
  scale.set_scale(calibration_factor);  // Sets the calibration factor.
  scale.wait_ready();
  reading = scale.get_units(10);        //Read scale in g/Kg
  raw = scale.read_average(5);          //Read raw value from scale too

  if (reading < 0) {
    reading = 0.00;     //Sets reading to 0 if it is a negative value, sometimes loadcells will drift into slightly negative values
  }

  String value_str = String(reading);
  String value_raw_str = String(raw);

  msg = "peso/";
  msg += value_str.c_str();
  mesh.sendSingle(to, msg);

  msg = "valuerawstr/";
  msg += value_raw_str.c_str();
  mesh.sendSingle(to, msg);
  scale.power_down();

}

void update_status()
{
  long uptime = millis() / 60000L;
  msg = "uptime/";
  msg += uptime;
  mesh.sendSingle(to, msg);
  msg = "nodeid/";
  msg += mesh.getNodeId();
  mesh.sendSingle(to, msg);
  msg = "freememory/";
  msg += String(ESP.getFreeHeap());
  mesh.sendSingle(to, msg);
  msg = "version/";
  msg += VERSION;
  mesh.sendSingle(to, msg);
  msg = "root/";
  msg += root_id;
  mesh.sendSingle(to, msg);
  msg = "ip/";
  msg += WiFi.localIP().toString();
  mesh.sendSingle(to, msg);
  msg = "wifisignal/";
  msg += String(WiFi.RSSI());
  mesh.sendSingle(to, msg);
  
}

void setup() {
  Serial.begin(115200);
  scale.begin(LOADCELL_DOUT_PIN, LOADCELL_SCK_PIN);   // Start scale on specified pins
  scale.wait_ready();                                 //Ensure scale is ready, this is a blocking function
  scale.set_scale();                                  
  scale.wait_ready();
  //scale.tare();                                       // Tare scale on startup
  //scale.wait_ready();

  //mesh.setDebugMsgTypes( ERROR | MESH_STATUS | CONNECTION | SYNC | COMMUNICATION | GENERAL | MSG_TYPES | REMOTE ); // all types on
  mesh.setDebugMsgTypes( ERROR | STARTUP | DEBUG );  // set before init() so that you can see startup messages

  mesh.init( MESH_PREFIX, MESH_PASSWORD, &userScheduler, MESH_PORT, WIFI_AP_STA, 11 );

  mesh.initOTAReceive(ROLE);
  mesh.setContainsRoot(true);
  mesh.setName(nodeName);
  mesh.onReceive(&receivedCallback);
  mesh.onNewConnection(&newConnectionCallback);
  mesh.onChangedConnections(&changedConnectionCallback);
  mesh.onNodeTimeAdjusted(&nodeTimeAdjustedCallback);

  userScheduler.addTask( taskSendMessage );
  userScheduler.addTask( taskSendMessage1 );
  taskSendMessage.enable();
  taskSendMessage1.enable();

}

void loop() {
  // it will run the user scheduler as well
  mesh.update();
}
`

RobTillaart, 2024-12-22:
Thanks for reporting the issue,
Time is very limited at the moment and it will probably take a few weeks before I can investigate.
Sorry for the inconvenience.

RobTillaart, 2024-12-22:
(updated your post to improve readability)

In a quick look I see nothing that points to the HX711 library.

Have you tried running the examples?  (adjust pins where needed of course)
Do they fail too?

sanluca, 2024-12-22:
I tried the code again following the HX_is_ready example. I'm afraid that the HX library is slowing down the mesh loop, but I'm not sure

RobTillaart, 2024-12-22:
If you roll back the HX711 library, does the problem disappear?

RobTillaart, 2025-01-02:
@sanluca 
Any progress / new insights / roll back /... to report?

sanluca, 2025-01-03:
Here you are. I've run more tests with different versions. The issue doesn't seem to be with your library, but perhaps with painlessmesh and ArduinoJSON. However, this still needs to be thoroughly tested. I'm closing this now. Thank you and happy new year

closed 2025-01-03 by sanluca

RobTillaart, 2025-01-03:
Happy new year to you too,

Never used the mesh lib, but you could check how the code works without the DEBUG flag. 
DEBUG mode often adds extra code that can have side effects (seen this too often in projects)
(just a thought).

sanluca, 2025-01-04:
> Happy new year to you too,
> 
> Never used the mesh lib, but you could check how the code works without the DEBUG flag. DEBUG mode often adds extra code that can have side effects (seen this too often in projects) (just a thought).

It might be an option, I will try. Thanks again