RobTillaart/HX711 #59: Confused with changes in the HX711 calibration constants

classbproject, 2025-01-28:
Last year, I built a Nano project that uses the HX711 ADC to read two load cells. Based on the weight difference, it plays an audio file.

In the code below, the calibration constants were in the 400 range as shown in the code below. However, the library has been updated. During a recent calibration cycle, these constants now appear as decimals (see the calibration output at the end). 

Should I update my code to include these as decimals?
```
// scale constants.
const float calibrationCell1 = 432.997039;
const float calibrationCell2 = 421.606750;
const long offsetCell1 = 4294948297;
const long offsetCell2 = 4294445015;
```
The most recent calibration output is as shown below,
```
CALIBRATION
===========
remove all weight from the loadcell
and press enter

Determine zero weight offset
OFFSET: 4294437740

place a weight on the loadcell
enter the weight in (whole) grams and press enter
WEIGHT: 150
SCALE:  0.524167

use scale.set_offset(4294437740); and scale.set_scale(0.524167);
in the setup of your project
```

RobTillaart, 2025-01-28:
> Should I update my code to include these as decimals?

Yes, you can as the calibration and the internal code can handle / use those decimals.
In fact not using the decimals would be a systematic error, although a small one
Note that rounding or truncating the value will give an error of max 1 in 432 in your case.
This would be roughly less than 0.25% so not that big of an error.

If you want you could round the value, 432.997039  would previously be 432 but actual 433 seems to be a better value to use.
Rounding would reduce the systematic error to 0.5 in 432 => ~ 0.13% so in the range of the accuracy of most loadcells.

classbproject, 2025-01-28:
Hi, Just so that I've understood this properly, its okay to use `scale.set_scale(0.524167);` instead of `scale.set_scale(433);`?

classbproject, 2025-01-28:
> Or using another zero reference?

Not sure I understand what you mean by this. Its the same loadcell. I was surprised by the difference as well and assumed that it is because of the library updates.

RobTillaart, 2025-01-28:
You can't break the setup so just give it a try if it gives meaningful outcomes.

Q: which versions did you compare exactly? So I can investigate.

classbproject, 2025-01-28:
The constants mentioned above were with version 0.5.0. The ones now are with the latest version.

RobTillaart, 2025-01-28:
Ok, I am going to have a quick look at the changes.

RobTillaart, 2025-01-28:
Changes between 0.5.0 and 0.5.2 (latest)

1) internal long _offset => int32_t _offset + get/set_offset() function
1) internal _lastRead => _lastTimeRead  + calling function

HX_calibration.ino
1) uint32_t offset => int32_t offset as it was possible to have a negative offset. This negative offset did not change the working of the library. See #54 - as adding e.g an offset of 4294811764 or -155532 gives the same effect (due to roll over math).

So in short I cannot explain your results from the changes in the code (yet, will check deeper as I have no setup)

----

Using the same weight:
Can you run the calibration sketch with version 0.5.0 and post the whole output.
Can you run the calibration sketch with version 0.5.2 and post the whole output.

RobTillaart, 2025-01-28:
> > Or using another zero reference?
> 
> Not sure I understand what you mean by this. Its the same loadcell. I was surprised by the difference as well and assumed that it is because of the library updates.

Did you call the **tare()** function before calibration? 
You must know the signal the load cell gives when it is in its zero position (normally this is no weight).

classbproject, 2025-01-28:
> Using the same weight:
> Can you run the calibration sketch with version 0.5.0 and post the whole output.
> Can you run the calibration sketch with version 0.5.2 and post the whole output.

Will do. 

> Did you call the **tare()** function before calibration?
> You must know the signal the load cell gives when it is in its zero position (normally this is no weight).

I used the calibration sketch from the examples. I believe it does call `tare()` before getting the existing offset.

classbproject, 2025-01-28:
I declared the calibration constants as a float. Is that the right data type for it, or do I need to use something else?

classbproject, 2025-01-29:
So I tried this again right now with 0.5.0 and 0.5.2 and now I'm getting consistent constants in the range of 422 and 435. Not sure why I was seeing values less than 1 yesterday. My apologies for being a bother.

RobTillaart, 2025-01-29:
No problem, 
to hear it all is stable and working is better news than having a serious bug :)

Assuming there are no related questions I close this issue.

closed 2025-01-29 by RobTillaart