RobTillaart/HX711 #60: Calibration units ? Float or int

mitra42, 2025-04-10:
Hi Rob

I'm confused by 
```
void calibrate_scale(uint16_t weight, uint8_t times = 10)
```

If I want to calibrate say a 2kg load_cell with a 1.5kg weight 

What value should I put here. 

The output is going to be a float, so I'm presuming the weight here should be the actual result I want i.e. `1.5`

Looking at the code 
```
void HX711::calibrate_scale(uint16_t weight, uint8_t times)
{
  _scale = (1.0 * weight) / (read_average(times) - _offset);
}
```
This makes sense, I just don't see why `weight` is a `uint16_t` rather than a `float`

RobTillaart, 2025-04-10:
Try 1500 grams

RobTillaart, 2025-04-10:
Try 1500 grams

RobTillaart, 2025-04-10:
@mitra42 
Have given the float parameter some thoughts and it increases functionality while not affecting the code size.

Snippet from the new readme.md 

----

- **void calibrate_scale(float weight, uint8_t times = 10)** 
The calibration weight is measured times x and averaged.
Weight is typical in grams, however any unit can be used.
Be aware this unit will also be returned by **get_units()**.

Since 0.5.3 the weight is defined as float which allows easier calibration in
other units e.g. define the weight as 2.5 kg instead of 2500 gram.
Also by using a float the range of calibration weights is substantially increased.
One can now define 250 gram as 250000 milligram, where before the max value 
was 65535 units (theoretical increase of precision from 4.8 to 6.9 digits).
This allows the calibration of superheavy load cells, e.g 500 kg and use a 
defined weight of e.g. 125.014 kg (e.g. a known volume of water).

Note: calibrate_scale() uses averaging and does not use the mode set.

Note: calibrate_scale() can have a negative value as weight e.g. force of a balloon.

RobTillaart, 2025-04-10:
@mitra42 
Created develop branch and a PR 
Expected to be merged in a few days.

RobTillaart, 2025-04-10:
Yes scale factors can be negative.

A load cell measures a force in a direction. They are mostly used for the force of gravity and the direction is + 

2 cases as example
1) imagine a project testing the lifting power of drones. These would be used in the other direction.
2) imagine someone (at least I did) who accidentally flipped the load cell during construction. It would also result in a negative scale factor.

There are more projects that can benefit from using a negative scale. It just tells us about the direction the force is heading.

mitra42, 2025-04-11:
Thanks - so in a "normal" configuration (they are weighing stuff), this shouldn't happen.

RobTillaart, 2025-04-11:
> I wonder if it would happen if positive and negative was reversed in the wiring ,

From Sparkfun:site

If the readings from the HX711 are opposite of what you are expect (for example the values decrease as you increase weight) simply reverse the O+/O- wires.

commit 5601e1fe (2025-04-11): fix #60, change calibration weight to float (#62)

- fix #60, change parameter **void calibrate_scale(float weight, uint8_t times = 10)**
- update readme.md
closed 2025-04-11 by RobTillaart