RobTillaart/HX711 #63: 80SPS on HX711?

mato295, 2025-04-16:
Hi Robert,

Im using your library with HX711 on my diploma thesis. Everything works well. Just for my application in some cases the 10SPS rate is not enough. I would like to use the 80SPS one. From datasheet for HX711 is clear that to obtain the 80SPS rate you need to unsolder the rate pin from GND and solder it to VCC to set it to 1(Rate = 1). But my question is if I'm gonna do it, will the library and mostly the reading functions work? As it is faster sample frequency than for 10 SPS. Or  Do I need to change something in the code of functions? 

Thank you a lot for your answer,
Take care
Martin

RobTillaart, 2025-04-16:
Hi Martin,
The library should work although I have never tested 80 SPS myself.

You could connect the RATE pin to a IO pin to be able to switch between the two modi. Could be useful as I recall the 10 SPS has a few bits more accuracy.
In theory 80vs10 could be 3 bits almost an extra decimal.

beniseman, 2025-04-17:
@mato295 the HX711 library works without issue at 80 Hz. The only change I made in my fork for the HX71708 was setting the data rate, and I have not encountered problems at 320 Hz. You won't find breakout boards for the HX71708 though, so you'd have to get an SOP breakout or make a custom pcb if you want to use that chip. 

The project that led me to try the HX71708 also led me to test the ADS1220 from Texas Instruments which goes up to 2kHz and is available on a breakout board.

mato295, 2025-04-18:
> Have you read this in the readme?
> 
> If you need more SPS you could consider using the HX71708 device. This is a close "relative" of the HX711 that allows to set the SPS to 10, 20, 80, or **320** Hz.
> 
> Link is in the (latest) readme.md

Yeah I saw it. But I didnt find any breakout board for it. And i dont have time to make my own PCB. Im completely satisfied with HX711, just need a bit more data with for just few seconds of measuring the force with Load cell.

mato295, 2025-04-18:
> [@mato295](https://github.com/mato295) the HX711 library works without issue at 80 Hz. The only change I made in my fork for the HX71708 was setting the data rate, and I have not encountered [...]

So, you just soldered the RATE pin of the HX711 to VCC to set it to 80 SPS, and everything works fine? Including the data read functions?
I was just wondering — since the sample rate is higher, could there be any issues with timing? Specifically, does DOUT still behave reliably at 80 SPS, and is there a risk of missing data if the microcontroller doesn’t check DOUT fast enough?

I also looked into the ADS1220, but I’ve already bought the HX711 for my project, and if I can get those 80 samples per second working reliably, it’ll be perfect.

RobTillaart, 2025-04-18:
FYI,

At RATE=80SPS the input noise is specified higher 

[image]

#### Clock Source Options (from datasheet)
_By connecting pin XI to Ground, the on-chip oscillator is activated. The nominal output data
rate when using the internal oscillator is 10 (RATE=0) or 80SPS (RATE=1)._

_If accurate output data rate is needed, crystal or external reference clock can be used. A crystal
can be directly connected across XI and XO pins. An external clock can be connected to XI pin,
through a 20pF ac coupled capacitor. This external clock is not required to be a square wave.
It can come directly from the crystal output pin of the MCU chip, with amplitude as low as 150 mV._

_When using a crystal or an external clock, the internal oscillator is automatically powered down._

@mato295 
What is the measurement range / accuracy / precision you need?

beniseman, 2025-04-18:
@mato295 the most common breakout(s) have the RATE pin tied to ground under the chip, so you can't just solder to VCC because you'll short it. You either need to desolder the entire chip and cut the trace, or bend the pin up before soldering a wire to ground.

beniseman, 2025-04-18:
The signalling works reliably at 80 Hz and at 320 Hz on the HX71708. Depending on how you structure your code you will drop below 80 SPS. The project I'm collaborating on samples a load cell, and outputs to a display and records logs to a microSD card. We are using ESP32S3.

With code structured as a loop that checks: `if (loadcell.is_ready()) {` and appending to microSD on every single sample:

HX711 at 80 Hz
> Average Sample Rate: 49.8680528802422
Median Sample Rate: 34.48275862068966
Standard Deviation of Sample Rate: 21.41728448075891
Minimum Sample Rate: 16.949152542372882
Maximum Sample Rate: 83.33333333333333

HX71708 at 320 Hz:
> Average Sample Rate: 99.7636758231504
Median Sample Rate: 100.0
Standard Deviation of Sample Rate: 9.369887440980351
Minimum Sample Rate: 17.54385964912281
Maximum Sample Rate: 142.85714285714286

 HX71708 at 320 Hz. Loop structure for sampling but batched writing to microSD via a FreeRTOS queue and `xTaskCreatePinnedToCore`. Note that the standard deviation is quite high. This occurred around batched microSD writes.
> Average Sample Rate: 340.1847977490912
Median Sample Rate: 330.6878306878307
Standard Deviation of Sample Rate: 71.6884014306761
Minimum Sample Rate: 198.68865487780647
Maximum Sample Rate: 930.2325581395348

ADS1220 at 2 kHz with nothing in the loop except button press handling. Sampling, display, and write to microSD all scheduled tasks. Specific timing for sampling and display via `vTaskDelayUntil(&lastWakeTime, pdMS_TO_TICKS(dataDelayMs));`

> Average Sample Rate: 1928.18
Median Sample Rate: 1926.78
Standard Deviation of Sample Rate: 6.50
Minimum Sample Rate: 1893.94
Maximum Sample Rate: 2123.14

beniseman, 2025-04-18:
Getting a max sample rate of 930 Hz on a chip rated for 320 might seem to contradict my comment that signalling is reliable. I believe that to be an artifact of having sampling and writing to sd on the same core. My understanding is that SD writes can still be in states where they are blocking, even if assigned to a lower priority task, and I think this was causing the timestamps I was applying to be unreliable. I would get an extremely high alternating with extremely low sample rate during microSD writes, up to 5500Hz on the ADS1220.

RobTillaart, 2025-04-18:
> and I think this was causing the timestamps I was applying to be unreliable

Compiler optimization, could also be a cause, 
you might make the timestamp variables volatile to exclude this cause.

mato295, 2025-04-23:
> [@mato295](https://github.com/mato295) the most common breakout(s) have the RATE pin tied to ground under the chip, so you can't just solder to VCC because you'll short it. You either need to desolder the entire chip and cut the trace, or bend the pin up before soldering a wire to VCC.

On my breakout board it looks like you just need to unsolder the rate pin of HX711 and solder it to VCC. Like there is nothing under the chip. Found the video where the guy have the same problem and he just solderes the Rate pin to VCC with 10kOhm resistor. Link for the video: https://youtu.be/uvn-J8CbtzM?si=GOA8AtHQqhWnDhma&t=521 . And I have the same breakout board. So I assume just by doing this I could obtain the 80 SPS sample rate. The only think I was curious about was if the library would work. So you said it is working without issue at 80SPS right?

beniseman, 2025-04-23:
> On my breakout board it looks like you just need to unsolder the rate pin of HX711 and solder it to VCC. Like there is nothing under the chip. Found the video where the guy have the same problem and [...]

If you desolder the pin it becomes irrelevant whether the ground connection is under the chip or in front of the pin. I mentioned it only because one of the people I'm collaborating with just soldered rate to vcc on the assumption that it had been left floating because no trace was visible.

>  The only think I was curious about was if the library would work. So you said it is working without issue at 80SPS right?

Yes.

Can you indulge my curiosity and share the topic of your thesis?

mato295, 2025-04-24:
So I tried it out today and soldered the Rate pin to VCC by a 4k7 SMD resistor. Everything works without any problem. Thank you for help guys! Attaching the photo of jumped Rate pin to 1.
[image]

closed 2025-04-24 by mato295