RobTillaart/HX711 #64: "Port closed: io: read/write on closed pipe" on ESP32S3 + FreeRTOS

andrei-zz, 2025-06-01:
Hi Rob,

We are trying to use your library for our FreeRTOS project but it seems like `HX711::begin` fails and give the "Port closed: io: read/write on closed pipe" error, which causes the opened Serial monitor to abort.

- Host OS: macOS 15.4 (M1)
- HX711 Version: 0.6.0
- Board: ESP32-S3 DevKitC-1 (FQBN: esp32:esp32:esp32s3)
```
% arduino-cli version
arduino-cli  Version: 1.2.0 Commit: Homebrew Date: 2025-02-24T15:52:58Z
```

<details>
  <summary>Here's the code we use for isolated testing</summary>

  ```cpp
  #include <FreeRTOS.h>
  #include <HX711.h>
  #include <math.h>
  
  // Pin definitions
  #define DOUT  21  // Connect to HX711 DOUT
  #define SCK   20  // Connect to HX711 SCK
  
  #define SCALE_COEFF 338.310394
  #define SCALE_OFFSET 1412837
  
  static HX711 scale;
  
  static void scaleSetup(void);
  static void Task_WEIGHT(void *pvParams);
  static double readWeight(void);
  
  void WEIGHT_init() {
    scaleSetup();
    xTaskCreatePinnedToCore(
      Task_WEIGHT, "Task_WEIGHT", 4096, NULL, 2, NULL, 1
    );
  
    Serial.printf("WEIGHT module initialized.\n\n");
  }
  
  static void scaleSetup() {
    Serial.printf("About to call scale.begin\n");
    scale.begin(DOUT, SCK, true);
    Serial.printf("scale.begin done\n");
  
    scale.set_offset(SCALE_OFFSET); 
    Serial.printf("scale.set_offset done\n");
    scale.set_scale(SCALE_COEFF);
    Serial.printf("scale.set_scale done\n");
  }
  
  static void Task_WEIGHT(void *pvParams) {
    for (;;) {
      Serial.printf("%.2f\n", readWeight());
      vTaskDelay(pdMS_TO_TICKS(200));
    }
  }
  
  static double readWeight() {
    if (scale.is_ready()) {
      return scale.get_units();
    } else {
      return nan("");
    }
  }
  
  
  
  void setup() {
    Serial.begin(115200);
    while (!Serial);
    Serial.printf("\n\n");
  
    WEIGHT_init();
  }
  
  void loop() {
  }
  ```
</details>

<details>
  <summary>Console output</summary>

  ```
  % arduino-cli board list
  Port                            Protocol Type              Board Name          FQBN                     Core
  /dev/cu.Bluetooth-Incoming-Port serial   Serial Port       Unknown
  /dev/cu.debug-console           serial   Serial Port       Unknown
  /dev/cu.usbmodem101             serial   Serial Port (USB) ESP32 Family Device esp32:esp32:esp32_family esp32:esp32
  /dev/cu.wlan-debug              serial   Serial Port       Unknown

  % make cupmon
  arduino-cli cache clean
  arduino-cli compile --fqbn esp32:esp32:esp32s3:CDCOnBoot=cdc   .
  Sketch uses 322674 bytes (24%) of program storage space. Maximum is 1310720 bytes.
  Global variables use 20684 bytes (6%) of dynamic memory, leaving 306996 bytes for local variables. Maximum is 327680 bytes.
  arduino-cli upload --fqbn esp32:esp32:esp32s3:CDCOnBoot=cdc   -p /dev/cu.usbmodem101 .
  esptool.py v4.8.1
  Serial port /dev/cu.usbmodem101
  Connecting...
  Chip is ESP32-S3 (QFN56) (revision v0.2)
  Features: WiFi, BLE, Embedded PSRAM 8MB (AP_3v3)
  Crystal is 40MHz
  MAC: 24:ec:4a:0e:bd:60
  Uploading stub...
  Running stub...
  Stub running...
  Changing baud rate to 921600
  Changed.
  Configuring flash size...
  Flash will be erased from 0x00000000 to 0x00004fff...
  Flash will be erased from 0x00008000 to 0x00008fff...
  Flash will be erased from 0x0000e000 to 0x0000ffff...
  Flash will be erased from 0x00010000 to 0x0005efff...
  Compressed 20208 bytes to 13057...
  Writing at 0x00000000... (100 %)
  Wrote 20208 bytes (13057 compressed) at 0x00000000 in 0.4 seconds (effective 434.5 kbit/s)...
  Hash of data verified.
  Compressed 3072 bytes to 146...
  Writing at 0x00008000... (100 %)
  Wrote 3072 bytes (146 compressed) at 0x00008000 in 0.1 seconds (effective 389.6 kbit/s)...
  Hash of data verified.
  Compressed 8192 bytes to 47...
  Writing at 0x0000e000... (100 %)
  Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.1 seconds (effective 571.2 kbit/s)...
  Hash of data verified.
  Compressed 322816 bytes to 173780...
  Writing at 0x00010000... (9 %)
  Writing at 0x0001c11c... (18 %)
  Writing at 0x00028cc8... (27 %)
  Writing at 0x0002e38a... (36 %)
  Writing at 0x00033d9b... (45 %)
  Writing at 0x000396ff... (54 %)
  Writing at 0x0003ed06... (63 %)
  Writing at 0x00044827... (72 %)
  Writing at 0x0004a15c... (81 %)
  Writing at 0x00054e9a... (90 %)
  Writing at 0x0005b14a... (100 %)
  Wrote 322816 bytes (173780 compressed) at 0x00010000 in 2.8 seconds (effective 915.1 kbit/s)...
  Hash of data verified.
  
  Leaving...
  Hard resetting with RTC WDT...
  New upload port: /dev/cu.usbmodem101 (serial)
  arduino-cli monitor -p /dev/cu.usbmodem101 --fqbn esp32:esp32:esp32s3:CDCOnBoot=cdc   --config baudrate=115200
  Monitor port settings:
    baudrate=115200
    bits=8
    dtr=on
    parity=none
    rts=on
    stop_bits=1
  
  Connecting to /dev/cu.usbmodem101. Press CTRL-C to exit.

  About to call scale.begin
  Port closed: io: read/write on closed pipe
  ```
</details>

However, when I switch from the USB port to UART and replace `Serial` with `Serial0`, the isolated sketch works. But the rest of my project doesn't work very well with the UART port.

<details>
  <summary>Console output for UART</summary>

  ```
  % arduino-cli board list
  Port                            Protocol Type              Board Name FQBN Core
  /dev/cu.Bluetooth-Incoming-Port serial   Serial Port       Unknown
  /dev/cu.debug-console           serial   Serial Port       Unknown
  /dev/cu.usbserial-10            serial   Serial Port (USB) Unknown
  /dev/cu.wlan-debug              serial   Serial Port       Unknown
  
  % make cupmon PORT=/dev/cu.usbserial-10
  arduino-cli cache clean
  arduino-cli compile --fqbn esp32:esp32:esp32s3:CDCOnBoot=cdc   .
  Sketch uses 331370 bytes (25%) of program storage space. Maximum is 1310720 bytes.
  Global variables use 20692 bytes (6%) of dynamic memory, leaving 306988 bytes for local variables. Maximum is 327680 bytes.
  arduino-cli upload --fqbn esp32:esp32:esp32s3:CDCOnBoot=cdc   -p /dev/cu.usbserial-10 .
  esptool.py v4.8.1
  Serial port /dev/cu.usbserial-10
  Connecting....
  Chip is ESP32-S3 (QFN56) (revision v0.2)
  Features: WiFi, BLE, Embedded PSRAM 8MB (AP_3v3)
  Crystal is 40MHz
  MAC: 24:ec:4a:0e:bd:60
  Uploading stub...
  Running stub...
  Stub running...
  Changing baud rate to 921600
  Changed.
  Configuring flash size...
  Flash will be erased from 0x00000000 to 0x00004fff...
  Flash will be erased from 0x00008000 to 0x00008fff...
  Flash will be erased from 0x0000e000 to 0x0000ffff...
  Flash will be erased from 0x00010000 to 0x00060fff...
  Compressed 20208 bytes to 13057...
  Writing at 0x00000000... (100 %)
  Wrote 20208 bytes (13057 compressed) at 0x00000000 in 0.5 seconds (effective 326.6 kbit/s)...
  Hash of data verified.
  Compressed 3072 bytes to 146...
  Writing at 0x00008000... (100 %)
  Wrote 3072 bytes (146 compressed) at 0x00008000 in 0.1 seconds (effective 223.5 kbit/s)...
  Hash of data verified.
  Compressed 8192 bytes to 47...
  Writing at 0x0000e000... (100 %)
  Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.2 seconds (effective 431.1 kbit/s)...
  Hash of data verified.
  Compressed 331520 bytes to 179502...
  Writing at 0x00010000... (9 %)
  Writing at 0x0001c28f... (18 %)
  Writing at 0x00028f1f... (27 %)
  Writing at 0x0002e619... (36 %)
  Writing at 0x00033fc1... (45 %)
  Writing at 0x000399a0... (54 %)
  Writing at 0x0003f0c9... (63 %)
  Writing at 0x00044a7c... (72 %)
  Writing at 0x0004a4b0... (81 %)
  Writing at 0x00054789... (90 %)
  Writing at 0x0005af3e... (100 %)
  Wrote 331520 bytes (179502 compressed) at 0x00010000 in 3.3 seconds (effective 799.5 kbit/s)...
  Hash of data verified.
  
  Leaving...
  Hard resetting via RTS pin...
  New upload port: /dev/cu.usbserial-10 (serial)
  arduino-cli monitor -p /dev/cu.usbserial-10 --fqbn esp32:esp32:esp32s3:CDCOnBoot=cdc   --config baudrate=115200
  Monitor port settings:
    baudrate=115200
    bits=8
    dtr=on
    parity=none
    rts=on
    stop_bits=1
  
  Connecting to /dev/cu.usbserial-10. Press CTRL-C to exit.
  -2994.55
  -1338.01
  24.55
  25.01
  24.89
  25.05
  ...
  ```
</details>

I'm unsure whether the issue is with the HX711 board, ESP32, or the library because your `.begin` code is quite simple. But since your "purely Arduino" examples and other sketches work fine, I suspect that the scheduling done by FreeRTOS might have messed up the timing or something that your library requires to work. What do you think? Let me know if you need more information.

Thank you in advance for your help,
Andrei

RobTillaart, 2025-06-01:
Thanks for the issue, 
Will dive into it coming week if time permits.
For now labelled it as question.

RobTillaart, 2025-06-01:
Not investigated yet but there are a few thoughts I would try 
- change the pins used for DOUT and CLK. 
- add a 5 second delay in start of setup() to give RTOS all time it need for whatever
- remove the keyword static before the constructor.
- manually set the pin mode and default value of  DOUT and CLK pin in setup().
- comment the delayMicroseconds in powerDown(). Might interfere with rtos timing
- try with a spare processor to (99%) exclude an ESP hardware related cause.

RobTillaart, 2025-06-01:
Might need to replace the delays in the library code with an rtos delay vTaskDelay(pdMS_TO_TICKS(nnn));

RobTillaart, 2025-06-01:
Do you have an old PC or laptop to test as terminal?  (to exclude a MacOS cause)

andrei-zz, 2025-06-01:
I will go ahead and try your suggestions later today, the school lab has some PCs that we can try stuff on.

andrei-zz, 2025-06-03:
We are able to get the code working with a different set of pins (1 and 2). Sorry for opening a non-issue.

RobTillaart, 2025-06-03:
@andrei-zz 

Good to hear it is solved!

> Sorry for opening a non-issue.

On contrary, there was an issue and there is a solution found, so useful for other people encountering the same error message.
The fact that the solution is "relative simple" in hindsight does not make it a non-issue.