RobTillaart/HX711 #67: new version

JustRollingAlong, 2025-06-23:
Hi Rob,

   Thank you for your code and the time you have pout into into it. :)
I am using it to help chicken farmers in the country of Georgia and release 6.0 works well.
When I update to 6.1 my porgam stopped working. I tried manually adding 

pinMode(A0, INPUT_PULLUP);
  pinMode(A2, INPUT_PULLUP);

(I have two HX711 boards)

It works but takes a long time to initialise when no scale is conneted.

09:10:15.162 -> R4 RTC OK.
09:10:36.072 -> ***Scale 1 Initialised
09:10:38.913 -> ***Scale 2 Initialised

is there a time out I can change ?

Thank you,
Colin Pridham

RobTillaart, 2025-06-24:
The A0 and A2 are the data lines of the HX711s?

RobTillaart, 2025-06-24:
The output you show is a delay of 20 seconds for the first HX711 and 0.9 seconds for the second HX711.

Can you post the code that shows what is done in your setup?

RobTillaart, 2025-06-24:
The library does not have long delays in its code, only **wait_ready()** has delays configured by the user. 

The code you posted is not enough to analyze the cause of the delay. 
The calls used are mostly to non HX711 library functions so I cannot see what they do.

There is a **delay(200)** in the code snippet seems to be not needed, but that does not explain the 2.8 seconds.

```
09:10:15.162 -> R4 RTC OK.
>>> ~20 seconds
09:10:36.072 -> ***Scale 1 Initialised
>>> ~2.8 seconds
09:10:38.913 -> ***Scale 2 Initialised
```

Can you post the whole code?  (attach as zip file)

JustRollingAlong, 2025-07-17:
> Hi
>
> Sorry for the delay. We have had family sickness to deal with.
>
> No need to apologize, hope all are well now.
>
> Just a question, does your code rely on pull up resistors to be on ?
>
> Depends on the length of your wires, if these are short (< 10 cm, 4 inch) it often will work without.
> It could give some bit flipping and thus faulty readings.
> For better signal quality pull ups are advised
>
> A very good discussion about pull ups can be found - https://www.gammon.com.au/forum/?id=10896 <https://www.gammon.com.au/forum/?id=10896&reply=5#reply5> &reply=5#reply5
> It is about I2C but just have a look at the scope pictures.

Thanks Rob. Much appreciated.