RobTillaart/HX711 #68: License violation

omani, 2025-09-05:
The new project should include the original copyright notice and license text, usually in a LICENSE or NOTICE file, or in the source code comments. The new project can add its own copyright notice for the new code, but it cannot remove or obscure the original.

The original copyright notice and license text must be included in the new project to comply with the MIT License. If they are not included, this is a violation of the license terms.

include this line to your MIT license:

```
MIT License

Copyright (c) 2018 Bogdan Necula
```

and then you can include your name on top of it.

RobTillaart, 2025-09-05:
Thanks for this issue, 
Labelled it as bug to give it priority. 
In my readme I refer to his work and the differences, 
Will update the license asap

RobTillaart, 2025-09-06:
@omani 

Created a develop branch which include the license fix and other changes. 
Will create a PR + merge later today

RobTillaart, 2025-09-06:
@omani 

FYI,
Also created develop branch / PR for HX711_MP and HX710AB which are derived from my HX711 library.
These also have function names used by Brogde.